hyzhou404/HUGSIM #1: Questions About Exposure Modeling in Paper

2747391052, 2024-12-16:
Dear Author,
Firstly, thanks for your Excellent work. I have some questions regarding the exposure modeling mentioned in your paper. 
My understanding is that the 3D Gaussian splatting method aims to reconstruct the original scene as accurately as possible. However, since the exposure settings of the original image are fixed, I am curious about how the affine transformation of the Gaussian colors can restore more realistic lighting conditions. Additionally, could you clarify how the specific parameters for the color affine transformation are determined? looking forward to your reply and guidance.
Sincerely,
...
[image]

hyzhou404, 2024-12-16:
The implementation can also be found in HUGS (https://github.com/hyzhou404/HUGS/blob/main/gaussian_renderer/__init__.py#L155). We estimate a subtle linear color transformation based on the camera poses, which can improve the PSNR.

hi-zgl, 2024-12-16:
> The implementation can also be found in HUGS (https://github.com/hyzhou404/HUGS/blob/main/gaussian_renderer/__init__.py#L155). We estimate a subtle linear color transformation based on the camera poses, which can improve the PSNR.该实现也可以在 HUGS （ https://github.com/hyzhou404/HUGS/blob/main/gaussian_renderer/__init__.py#L155 中找到）。我们根据摄像机姿势估计细微的线性颜色转换，这可以提高 PSNR。

Okay, I will check the relevant code. Thank you for your reply.

closed 2024-12-16 by hi-zgl