hyzhou404/HUGSIM #2: About the non-native car insertion

zzz5y, 2024-12-20:
Hello! Thank you for your excellent work and open source!
I am curious about the insertion of non-native cars!
As far as I know, it is based on 3dgs on 3Drealcar. So in the code, is it implemented by inserting the ply of 3dgs non-native cars (which contains many Gaussian balls)?
Looking forward to your corrections and links to relevant codes
If the non-native car is inserted by inserting the ply of non-native cars trained by 3dgs, will there be a release of such a reconstruction result demo?
Thank you very much for your excellent work!

你好！感谢你的出色工作和开源！
我很好奇非原生汽车的插入！
据我了解，它基于 3Drealcar 上的 3dgs。因此实现在代码里，就是通过插入 3dgs 非原生汽车的ply（里面包含很多高斯球）实现的吗
期望得到您的指正和相关代码的链接
如果是通过插入 3dgs 训练出来的非原生汽车的ply实现插入非原生汽车,请问会有这种重建结果demo的release吗
再次感谢您优秀的工作！

hyzhou404, 2024-12-20:
Hi, we integrate the non-native 3dgs models into reconstructed scenes using generated trajectories. The code of insertion is now available, the 3DRealCars reconstruction results, along with sample scenes and scenarios will be released next week.

zzz5y, 2024-12-20:
> Hi, we integrate the non-native 3dgs models into reconstructed scenes using generated trajectories. The code of insertion is now available, the 3DRealCars reconstruction results, along with sample scenes and scenarios will be released next week.

Got it！Thanks a lot!

closed 2024-12-20 by zzz5y