hyzhou404/HUGSIM #4: rasterization argument  error

ANaughtySnail, 2024-12-25:
The following error occurred when I ran the script ：train_ground.py . 
I checked that all versions of gsplat do not have this parameter. Is this missing parameter added by the author

Training progress:   0%|                                                              | 0/30000 [00:00<?, ?it/s]Traceback (most recent call last):
  File "/root/outside/HUGSIM/train_ground.py", line 244, in <module>
    main()
  File "/root/outside/HUGSIM/train_ground.py", line 240, in main
    training(cfg)
  File "/root/outside/HUGSIM/train_ground.py", line 77, in training
    render_pkg = render_ground(viewpoint_cam, gaussians, background)
  File "/root/outside/HUGSIM/gaussian_renderer/__init__.py", line 200, in render_ground
    renders, render_alphas, info = rasterization(
TypeError: rasterization() got an unexpected keyword argument 'smts'

Hope the author can reply me！ thanks

hyzhou404, 2024-12-25:
It seems you installed official version of gsplat and encountered the error. Instead, ours modified version of gsplat should be installed from the source code, available at https://github.com/hyzhou404/HUGSIM_splat/tree/main.

ANaughtySnail, 2024-12-26:
Thanks, this problem is solved by using modified version gsplat！

closed 2024-12-26 by hyzhou404