hyzhou404/HUGSIM #5: Sample data missing images

ANaughtySnail, 2024-12-26:
Hello, thank you for your excellent work, we are currently trying to run and learn from your work.
The following exception occurs when we use the sample data for reconstruction. 

  File "/root/outside/HUGSIM/train_ground.py", line 244, in <module>
    main()
  File "/root/outside/HUGSIM/train_ground.py", line 240, in main
    training(cfg)
  File "/root/outside/HUGSIM/train_ground.py", line 44, in training
    train_cams, test_cams, _ = load_cameras(cfg, cfg.data_type, True)
  File "/root/outside/HUGSIM/scene/__init__.py", line 19, in load_cameras
    scene_info = sceneLoadTypeCallbacks['HUGSIM'](args.source_path, data_type, ignore_dynamic)
  File "/root/outside/HUGSIM/scene/dataset_readers.py", line 187, in readHUGSIMInfo
    train_cam_infos, test_cam_infos, verts = readHUGSIMCameras(path, data_type, ignore_dynamic)
  File "/root/outside/HUGSIM/scene/dataset_readers.py", line 102, in readHUGSIMCameras
    image = imread(rgb_path)
  File "/usr/local/lib/python3.10/site-packages/imageio/core/request.py", line 408, in _parse_uri
    raise FileNotFoundError("No such file: '%s'" % fn)
FileNotFoundError: No such file: '/root/outside/HUGSIM/sample_data/sample_data/data/scene-0383/images/CAM_FRONT/00000.jpg'

We found the missing pictures in the data/scene-0383/images/CAM_FRONT_LEFT/.    Some images missing?

hyzhou404, 2024-12-26:
Hi, sorry for the missing files in the compressed folder. I'll re-upload them later.

HENUwjj, 2024-12-30:
hello,Thank you for sharing this facial recognition project! I’ve been trying to run the code.

Dataset Information: Could you clarify which dataset the image file is based on? It would be great if you could provide more details about the dataset or a link to where I can access similar data.

Missing Images: I noticed that some images seem to be missing, especially the ones required during the process. Would it be possible to upload the missing images or provide a link where they can be downloaded?

hyzhou404, 2024-12-30:
@ANaughtySnail @HENUwjj Hi, I have re-upload the files in https://huggingface.co/datasets/hyzhou404/HUGSIM/tree/main/sample_data. Please have a check.

hyzhou404, 2024-12-30:
> hello,Thank you for sharing this facial recognition project! I’ve been trying to run the code.
>
> Dataset Information: Could you clarify which dataset the image file is based on? It would be great if [...]

The sample sequence is scene-0383 from the nuScenes dataset. You can reproduce the data by following the instructions at https://github.com/hyzhou404/HUGSIM/blob/main/data/README.md.

closed 2025-01-07 by hyzhou404