hyzhou404/HUGSIM #6: Question about python environment

suminmin, 2024-12-30:
Thank you for sharing your excellent work!

I have a question regarding running the e2e. Initially, I attempted to install both HUGSIM and UniAD_SIM, but I couldn't get it to work. Upon further investigation, I came across the following line in the code:

[UniAD_SIM](https://github.com/hyzhou404/UniAD_SIM/tree/main)/[tools](https://github.com/hyzhou404/UniAD_SIM/tree/main/tools)/e2e.sh

`conda activate uniad`

Does this imply that two separate Python environments need to be prepared with Conda?

One for HUGSIM, using Python 3.11 with `torch==2.2.0`, and another for UniAD/VAD, using Python 3.8 with `torch==1.9.1`, `mmcv-full==1.4.0`, and `mmdet3d==0.17.1`.

hyzhou404, 2024-12-30:
Hello, as a simulator, we don't require client algorithms to use the same environment as ours. Therefore, we maintain separate Python environments for the simulator and the client. For UniAD, please refer to the official installation instructions, the specific package versions may differ from your CUDA environments. Since the official UniAD implementation does not provide direct inference code, we recommend using our modified version of UniAD available at https://github.com/hyzhou404/UniAD_SIM.

suminmin, 2024-12-30:
Thanks for the clarification!

closed 2024-12-30 by suminmin

rdesc, 2025-03-12:
Can this info maybe be added to the readme here https://github.com/hyzhou404/HUGSIM?tab=readme-ov-file#simulation ?