hyzhou404/HUGSIM #7: Question about Rendering FPS

hi-zgl, 2025-01-02:
Hello Author,

Thank you for your great work! I conducted experiments using the data you provided on a 4090 GPU to measure the FPS. Ideally, the FPS should be quite high, but I found that the calculated FPS is relatively low. 

Here is the method I used to calculate the FPS:

In the file [hug_sim.py](https://github.com/hyzhou404/HUGSIM/blob/main/sim/hugsim_env/envs/hug_sim.py), I modified the code to record the render time and added a class attribute `self.render_time` to store it. The specific code modification is as follows:

```python
def _get_obs(self):
    rgbs, semantics, depths = {}, {}, {}
    start_time = time.time()  # Start recording time
    v2front = self.cam_params['CAM_FRONT']["v2c"]
    for cam_name, params in self.cam_params.items():
        intrinsic, v2c = params['intrinsic'], params['v2c']
        c2front = v2front @ np.linalg.inv(v2c) @ self.cam_rect
        c2w = self.ego @ c2front
        viewpoint = create_cam(intrinsic, c2w)
        render_pkg = self.render_fn(viewpoint=viewpoint, prev_viewpoint=None, **self.render_kwargs)
        rgb = (torch.permute(render_pkg['render'].clamp(0, 1), (1,2,0)).detach().cpu().numpy() * 255).astype(np.uint8)
        smt = torch.argmax(render_pkg['feats'], dim=0).detach().cpu().numpy().astype(np.uint8)
        depth = render_pkg['depth'][0].detach().cpu().numpy()
        if (self.data_type == 'waymo' or self.data_type == 'kitti360') and 'BACK' in cam_name:
            rgbs[cam_name] = np.zeros_like(rgb)
            semantics[cam_name] = np.zeros_like(smt)
            depths[cam_name] = np.zeros_like(depth)
        else:
            rgbs[cam_name] = rgb
            semantics[cam_name] = smt
            depths[cam_name] = depth
    end_time = time.time()  # End recording time
    self.render_time = end_time - start_time  # Save render time
    return {
        'rgb': rgbs, 
        'semantic': semantics,
        'depth': depths,
    }
```
Then, I added the render time to the info dictionary:
```python
info['render_time'] = self.render_time
```
In the file [closed_loop.py](https://github.com/hyzhou404/HUGSIM/blob/main/closed_loop.py), I calculated the FPS by averaging the render times as follows:
```python
    while not done:
        if obs is None or info is None:
            obs, info = env.reset()
            render_times.append(info['render_time'])  # Save render time
        print('ego pose', info['ego_pos'])

        with open(obs_pipe, "wb") as pipe:
            pipe.write(pickle.dumps((obs, info)))
        with open(plan_pipe, "rb") as pipe:
            plan_traj = pickle.loads(pipe.read())
        if plan_traj is not None:
            acc, steer_rate = traj2control(plan_traj, info)
            action = {'acc': acc, 'steer_rate': steer_rate}
            obs, reward, terminated, truncated, info = env.step(action)
            render_times.append(info['render_time'])  # Save render time
            cnt += 1
            done = terminated or truncated or cnt > 400
        else:  # AD Side Crushed
            observations_save.append(obs)
            infos_save.append(info)
            done = True

        imu_plan_traj = plan_traj[:, [1, 0]]
        imu_plan_traj[:, 1] *= -1
        global_traj = traj_transform_to_global(imu_plan_traj, info['ego_box'])
        save_data['frames'].append({
            'time_stamp': info['timestamp'],
            'is_key_frame': True,
            'ego_box': info['ego_box'],
            'obj_boxes': info['obj_boxes'],
            'obj_names': ['car' for _ in info['obj_boxes']],
            'planned_traj': {
                'traj': global_traj,
                'timestep': 0.5
            },
            'collision': info['collision'],
            'rc': info['rc']
        })
    # Calculate fps
    avg_render_time = sum(render_times) / len(render_times) if len(render_times) > 1 else None
    fps_render = 1 / avg_render_time if avg_render_time else None
```
**The calculated FPS is around 10.**
I would greatly appreciate it if you could help me understand why the FPS is lower than expected and if there are any potential code optimizations I could implement.
Thank you in advance for your time and assistance!

**Best regards**

hyzhou404, 2025-01-02:
Hi, that is a good question! In the paper, we measure FPS by only considering the time taken by rendering functions. I think the FPS is slowed down by operation like `create_cam` and `detach().cpu().numpy()`. These time consuming operations can be optimized with some extra engineering efforts. 
However, currently, the main bottleneck in closed-loop testing speed comes from the AD algorithms rather than the simulator itself (FPS of UniAD and VAD is below 5 FPS). We will look into optimizing the speed of simulation later.

hi-zgl, 2025-01-03:
> Hi, that is a good question! In the paper, we measure FPS by only considering the time taken by rendering functions `self.render_fn()`. I think the FPS is slowed down by operations like `create_cam()` and `detach().cpu().numpy()`. These time consuming operations can be optimized with some extra engineering efforts. However, currently, the main bottleneck in closed-loop testing speed comes from the AD algorithms rather than the simulator itself (FPS of UniAD and VAD is below 5 FPS). We will look into optimizing the speed of simulation later.

Thank you for your  response. I conducted further experiments, and as you pointed out, when only considering the time taken by the rendering function `self.render_fn()`, the FPS is significantly higher, averaging around 95 FPS. Additionally, I observed that the time spent on `detach().cpu().numpy()` is more than that spent on the `create_cam()` function.

Thanks again for your time and explanation! Wishing you happiness every day！🙂

closed 2025-01-03 by hi-zgl