hyzhou404/HUGSIM #8: The installation requirements of the Python environment are not clear.

zhude233, 2025-01-07:
This is a very good job, but when we tried to reproduce it, we found that the environmental dependencies were ambiguous and the installation was difficult. Could you provide a clear installation requirement?

hyzhou404, 2025-01-07:
Hi, we also notice the installation of environment can be challenging. Because some packages depend on CUDA environment which may vary across users' machines, and some packages have to be installed from the source code. 
One possible solution is to provide a Docker environment, but our server cannot connect to the DockerHUB or mirrors. We will try to create a one-click installation way in the future to simplify the process.

zhude233, 2025-01-08:
Perhaps you should test the installation process by yourselves and then write out the installation process. Excessive referencing of the github repositories of packages, this can easily lead to problems such as numpy version conflicts, and the readme in HUGSIM_splat has not been correctly modified.

zhude233, 2025-01-08:
In the case of installing in the order given in this readme, it is possible to encounter a compilation error problem caused by too high a version of numpy on HUGSIM_splat.

closed 2025-01-08 by zhude233

iwander-all, 2025-01-24:
environment is very hard to build, I have tried days and failed...

rdesc, 2025-03-12:
Any luck btw @iwander-all ? Maybe we can start a thread on the various fixes that were required and compile a single working `requirements.txt` file

hyzhou404, 2025-04-19:
I solved the problem by using pixi. Now you can install the environment by only two simple stages!