hyzhou404/HUGSIM #13: Closed loop simulation of hugsim when AD is uniad

sdc-sdd, 2025-03-25:
Hello, I am interested in the implementation of Uniad closed-loop simulation. What is the data format input to Uniad? The aforementioned data is stored in scene format, but I noticed that the configuration file in uniad_sim still uses nuscenes_infos_temporal_train. pkl, etc. Looking forward to your reply.

jkjls, 2025-03-31:
Thank you for your patient answer. How to use Uniad for open-loop evaluation in your method? I want to observe the difference between the Uniad test results of the reconstructed scene using 3D Gaussian sputtering and the original Nuscenes data for the same batch of data.

jkjls, 2025-03-31:
> 您好，nuscenes_infos_temporal_train.pkl 文件实际上在模拟过程中并未使用；它只是一个配置模板。官方的 UniAD 实现不包含专用的推理代码。我们对源代码进行了最小程度的更改以启用推理，同时保留了一些不会影响输出结果的模板配置。
> 
> 我们使用的实际输入可以在此链接中找到https://github.com/hyzhou404/UniAD_SIM/blob/5fb279e39912a5ac7f58e00d56b065cadcd0a749/tools/closeloop/e2e.py#L232。

Thank you for your outstanding work. How can Uniad be used for open-loop evaluation in your approach? I want to observe the difference between the Uniad test results of the reconstructed scene using 3D Gaussian sputtering and the original Nuscenes data for the same batch of data.

hyzhou404, 2025-03-31:
You can modify the input camera poses from the closed-loop simulation to use the real captured camera poses from the original datasets (eg. in lines around https://github.com/hyzhou404/HUGSIM/blob/main/sim/hugsim_env/envs/hug_sim.py#L218). We haven't provided a direct method for this testing, so it will require a few changes to our code.

> Thank you for your outstanding work. How can Uniad be used for open-loop evaluation in your approach? I want to observe the difference between the Uniad test results of the reconstructed scene using 3D Gaussian sputtering and the original Nuscenes data for the same batch of data.

closed 2025-04-11 by hyzhou404