hyzhou404/HUGSIM #16: project test data result not good

xiaoliangxue, 2025-04-30:
I just run train_ground.py with test data in project, but render result in test data not good. help check it
train log:
Training progress: 23%|███████████████████████▌ | 7000/30000 [05:30<13:59, 27.41it/s, RGB=0.0003, Semantic=0.0208, dist=0.1843]
[ITER 7000] Evaluating test: L1 0.0136 PSNR inf SSIM 0.9409 Lpips 0.0568

[ITER 7000] Evaluating train: L1 0.0132 PSNR inf SSIM 0.9436 Lpips 0.0558

[ITER 7000] Saving Checkpoint
[Open3D WARNING] Write Ply clamped color value to valid range
Training progress: 50%|██████████████████████████████████████████████████ | 15000/30000 [13:13<09:51, 25.38it/s, RGB=0.0253, Semantic=0.0213, dist=0.6210]
[ITER 15000] Evaluating test: L1 0.0119 PSNR inf SSIM 0.9448 Lpips 0.0500
Training progress: 50%|██████████████████████████████████████████████████ | 15000/30000 [13:26<09:51, 25.38it/s, RGB=0.0253, Semantic=0.0213, dist=0.6210]
[ITER 15000] Evaluating train: L1 0.0114 PSNR inf SSIM 0.9483 Lpips 0.0491

[ITER 15000] Saving Checkpoint
Training progress: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████| 30000/30000 [26:17<00:00, 19.02it/s, RGB=0.0142, Semantic=0.0208, dist=0.0293]

[ITER 30000] Evaluating test: L1 0.0093 PSNR inf SSIM 0.9513 Lpips 0.0413

[ITER 30000] Evaluating train: L1 0.0087 PSNR inf SSIM 0.9558 Lpips 0.0402

[ITER 30000] Saving Checkpoint

example result:
HUGSIM/hugsim_output/scene-0383/ground/save_test/CAM_BACK_LEFT_00009.png

[image]

hyzhou404, 2025-04-30:
The ground mask is weird, as it seems to cover most of the pixels. Did you generate the mask yourself, or did you make any modifications during the mask loading process? It would be a good idea to check the masked ground truth images before training.

xiaoliangxue, 2025-04-30:
when i first run the demo, with no any modifications, I found left/right camera render result has the problem, but front/rear camera  result is good.  By the way, where is the masked ground truth images in data path?

hyzhou404, 2025-05-07:
I reviewed the data, and the output you provided is accurate. The method only reconstructs the ground in the first stage. In back-left-0009, the majority of the areas are non-ground.

[image]

closed 2025-05-11 by hyzhou404