hyzhou404/HUGSIM #19: Non-Native Vehicle Reconstruction

zzz5y, 2025-05-06:
Hello! I wonder what kind of method do you use to reconstruction Non-Native Vehicle
I realize that you get the Non-Native Vehicle format is .pth instead of ply. Can you tell me the reason why not use .ply?
Thank you for your great job!

hyzhou404, 2025-05-06:
Hi, as mentioned in README, you can convert the pth file to ply by using `python eval_render/convert_vehicles.py --vehicle_path ${PATH_3DRealCar}` .

zzz5y, 2025-05-06:
> Hi, as mentioned in README, you can convert the pth file to ply by using `python eval_render/convert_vehicles.py --vehicle_path ${PATH_3DRealCar}` .

Hello sir.
Thank you for your quick reply.And Sorry for my rudeness.
And how do you reconstruct this 3DrealCar dataset? use basic 3dgs or other improved 3dgs?Thank you so much sir
Because i notice the result seems good after I turn the non natice vehicle .pth to ply.

hyzhou404, 2025-05-07:
The reconstruction method is almost the vanilla 3DGS. The primary distinctions are the inclusion of the alpha loss (as described in equation 13 of the paper) and the use of shadow Gaussians.

zzz5y, 2025-05-08:
> The reconstruction method is almost the vanilla 3DGS. The primary distinctions are the inclusion of the alpha loss (as described in equation 13 of the paper) and the use of shadow Gaussians.重建方法几乎是普通的3DGS。主要的区别是包含了alpha损失（如本文公式13所述）和使用阴影高斯函数。

Got it.Thank you sir

closed 2025-05-08 by zzz5y