hyzhou404/HUGSIM #28: why not use dynamic_mask in merge_depth_wo_ground.py

YinXunzhe, 2025-05-28:
Hi, I wonder why "mask = dynamic_mask & smt_mask"  is commented out in merge_depth_wo_ground.py ?

https://github.com/hyzhou404/HUGSIM/blob/47a0ac629172023e40f039a9703cc8abf31354d2/data/utils/merge_depth_wo_ground.py#L82

hyzhou404, 2025-06-05:
It is a mistake in the released version. I have fixed it in the new version. Thanks for pointing it out!

utc1205, 2025-06-18:
> Hi, thanks for your issue. This may be a mistake in the released version. I will check it later.
if using mask = dynamic_mask & smt_mask, it means the saved points3d.ply is only for dynamic objects. but i think that points3d.ply in merge_depth_wo_ground.py is used to save the pointcloud of the whole scene expect background pointcloud.  I'am a little confused, looking farward to your reply. Thanks.