hyzhou404/HUGSIM #32: About Lidar rendering

lnSong, 2025-06-27:
Thank you for your excellent work. I have a few questions:
1. Does Hugsim support LiDAR rendering?
2. Does Hugsim allow replacing its own 3DGS module? For example, replacing it with a 3DGS that can render LiDAR.

yzli-0910, 2025-06-27:
Same question for me.

It would be great if LiDAR rendering is supported so many multi-modal AD models can be tested as well.

hyzhou404, 2025-07-16:
We did not implement LiDAR rendering, as our focus is on constructing a vision-based simulator. However, since we can render depth maps from 3DGS, it would theoretically be possible to simulate LiDAR. But we do not have plans to implement this feature at this time.