hyzhou404/HUGSIM #36: Training Loss Fails to Converge in train_ground.py？

M21F-Liao, 2025-07-09:
Thank you for your excellent work! While reproducing this project, I noticed that several train_loss values in train_ground.py **failed to converge**. 
What could be the reasons for this issue? (I’m using the sample dataset Nuscenes-0383, default hyperparameters）

[image]

hyzhou404, 2025-07-09:
Hi, I have not encountered this situation during my experiments, particularly with the sample data nusc-0383, which has been thoroughly tested. Could you please provide more details, such as rendered images and any changes made to the code?

hyzhou404, 2025-07-10:
The results appear to be correct, and the PSNR is infinite due to a numerical error in the evaluation. I have also encountered numerical errors in my experiments, but they did not impact the overall results. I think you can proceed to the next step!

M21F-Liao, 2025-07-10:
> The results appear to be correct, and the PSNR is infinite due to a numerical error in the evaluation. I have also encountered numerical errors in my experiments, but they did not impact the overall results. I think you can proceed to the next step!

I think the raw input point cloud data(ground_points3d.ply) was already close enough to the ground truth, resulting in a very small initial loss, which is why the optimization process didn't have a noticeable effect. Thank you for your feedback!

closed 2025-07-10 by M21F-Liao